kingspp/RPi-AUI #22: Upgrade to GPLv3?

pickfire, 2015-01-05:
https://www.gnu.org/licenses/rms-why-gplv3.html

pickfire, 2015-01-10:
You have upgraded but you forgot to put the first line which is:
```
  To do so, attach the following notices to the program.  It is safest
to attach them to the start of each source file to most effectively
state the exclusion of warranty; and each file should have at least
the "copyright" line and a pointer to where the full notice is found.

    <one line to give the program's name and a brief idea of what it does.>
    Copyright (C) <year>  <name of author>

    This program is free software: you can redistribute it and/or modify

```

pickfire, 2015-01-10:
There is also a lot of trailing error which I got on vim on those files. Such as `<space> -> <dot>`:
```
This is an example..............
and this is another......
```

kingspp, 2015-01-12:
Change and commit

pickfire, 2015-01-12:
Why sometimes when I push, this error shows up?
```
To https://github.com/pickfire/dotfiles
 ! [rejected]        master -> master (non-fast-forward)
error: failed to push some refs to 'https://github.com/pickfire/dotfiles'
hint: Updates were rejected because the tip of your current branch is behind
hint: its remote counterpart. Merge the remote changes (e.g. 'git pull')
hint: before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.
```

kingspp, 2015-01-16:
You need to pull before you merge

pickfire, 2015-01-16:
I don't mean merge but I meant push.

kingspp, 2015-01-16:
Even for push you do the same

pickfire, 2015-01-16:
I thought `git pull` == `git fetch` + `git merge`?

kingspp, 2015-01-16:
If i have committed a change before you and you commit to it again without pull, you will get an error. Always pull the files before you start modifying.

pickfire, 2015-01-16:
Just `git fetch kingspp` or `git pull kingspp`?

kingspp, 2015-01-16:
git pull <branch>

pickfire, 2015-01-16:
solved

closed 2015-03-22 by kingspp