kingspp/RPi-AUI #24: Colours

pickfire, 2015-01-09:
Colored man pages with `less`
-------------------------------------

https://wiki.archlinux.org/index.php/Man_page#Colored_man_pages
https://github.com/robbyrussell/oh-my-zsh/blob/master/plugins/colored-man/colored-man.plugin.zsh

Colors for core utils
------------------------

https://wiki.archlinux.org/index.php/Core_utilities

kingspp, 2015-01-12:
What do you propose?

pickfire, 2015-01-12:
On 12/01/2015 16:27, Prathyush wrote:
> What do you propose?
Do some basic configuration for colors on core-utils.
Such as adding tips and tricks section on app.sh.

kingspp, 2015-01-16:
I will un-merge the changes, due to many conflicts and errors. You can download your previous changes here. The last pull request will be reset.
Your pull request with changes: https://github.com/kingspp/RPi-AUI/releases/tag/v6.2

pickfire, 2015-01-16:
Ok.

kingspp, 2015-01-16:
Review your changes. Test the script and create a pull request. I am using this for my current H/W project. So its high time that the script is error free.!!

pickfire, 2015-01-16:
H/W project?

pickfire, 2015-01-16:
Maybe let's split the branches like `git flow` that have a *develop* and *master* so that you doesn't worry about anything. Is it okay?

kingspp, 2015-01-16:
I am an Electronics student. I am implementing an idea using RPi. If that helps, then lets do that

pickfire, 2015-01-16:
I am still a student

closed 2015-03-22 by kingspp