kingspp/RPi-AUI #25: Look at this

pickfire, 2015-01-10:
https://github.com/altercation/archinit/blob/master/archinit

Do you think we should add this to our README.md for ideas?

pickfire, 2015-01-10:
This too:

https://github.com/altercation/atypical

pickfire, 2015-01-10:
And should we do the following?

1. move the archi.sh to RPi-AUI/archi.sh and not RPi-AUI/AUI/archi.sh
2. rename AUI to aui
3. add an app.sh in AUI to manage and install application
4. add help pages to archi.sh

you could put tick like this:

```
- [x] for a tick
- [ ] for no tick
```

kingspp, 2015-01-12:
It looks complicated

pickfire, 2015-01-12:
Add that to to-do list?

pickfire, 2015-01-16:
This site had good CONTRIBUTING.md except that we doesn't need that git flow as there is less people:
https://github.com/powerline/powerline

pickfire, 2015-01-18:
This [style guide](https://google-styleguide.googlecode.com/svn/trunk/shell.xml?showone=Indentation#Indentation, "google styleguide"). Okay?

kingspp, 2015-01-18:
For what purpose?

pickfire, 2015-01-18:
On 18/01/2015 21:16, Prathyush wrote:
> For what purpose?

Using 2 spaces for indentation would be better than using a tab.

kingspp, 2015-03-19:
I am using notepad++, tab space is said to be long, when it is said to be committed to the the github

closed 2015-03-19 by kingspp

pickfire, 2015-03-19:
What do you mean ***long***?

kingspp, 2015-04-04:
I mean it takes two tab spaces in github, but only a single tab in notepad++

pickfire, 2015-04-04:
How long is a single tab? I think it is almost the same for a tab and two spaces?