kingspp/RPi-AUI #29: Use `hostnamectl` for systemd

pickfire, 2015-02-17:
Do you think the *function* `hname` should be changed to what is said in the [wiki][Arch's Wiki] using the `systemd` method?
```bash
function hname() {
  echo "Your current hostname is:" $(hostname)
  $rpi_aui/./yn.sh "Do you wish to change the hostname? [y/N]" || ui
  cp /etc/hostname /etc/hostname.old # Create a backup for hostname
  rm /etc/hostname; echo -n "Enter the new hostname:"; read hn
  echo "$hn" > /etc/hostname; echo
  echo "Your new hostname is:" $(hostname)
}
```
And do you know what does `git-init-linux.sh` do?

[Arch's Wiki]: https://wiki.archlinux.org/index.php/Network_configuration#Set_the_hostname

kingspp, 2015-02-20:
well if it works, change it. Git-init-linux script was created to handle installation and configuration of a new Git install...

pickfire, 2015-02-20:
Yeah it works, I have add it on 6a5b8d2 commit in (my repository)[github.com/pickfire/RPi-AUI]. Should the `git-init-linux.sh` be moved into `AUI`? I currently have no idea how to work on `install.sh` and `uninstall.sh` so I will not have any update for that currently.

kingspp, 2015-02-21:
No. git-init-linux is a separate script used to initialize any git repo. It is not a part of AUI

kingspp, 2015-02-21:
I shall update you regarding those install and un-in scripts

pickfire, 2015-02-22:
I have made some changes for `install.sh` and `uninstall.sh` into a file call `setup.sh`. `setup.sh` can install and uninstall `aui` and help is provided with `--help`, and when no parameters is given `setup.sh` will install by default. Maybe this should be added to *README.md* to show how to install?

Although I have change a bit of `setup.sh` but I haven't change the `oc`, `aui`...And maybe some should be merged into `archi.sh`. Should we add a variable to `setup.sh` to specified where to install if someone else wanted to install at other directories such as */bin* or */usr/sbin*?

kingspp, 2015-03-19:
As for arch Linux, the rule is to install in /opt directory, so I chose that path.

pickfire, 2015-03-19:
I have made a `test.sh` for testing for a new table in `oc.sh` try testing out. And should the core installation copy or just create a soft link for the applications?

kingspp, 2015-04-04:
I did not get what do you men by core/ soft link?

pickfire, 2015-04-04:
I mean to just `cp -f file /usr/bin/file` instead of `ln -s file /usr/bin/file` as someone will accidentaly delete the files, so I need to think what we need to do about `main.sh`.

kingspp, 2015-06-27:
Any progress?

pickfire, 2015-06-28:
I did not work on this, just working on the table which I had no idea how to continue (test.sh), but seldom work on RPi-AUI now.

kingspp, 2015-07-01:
Getting regular Arch updates ?

pickfire, 2015-07-01:
On Wed, Jul 01, 2015 at 11:52:58AM -0700, Prathyush wrote:
>Getting regular Arch updates ?

What do you mean?

-- 
 _____________________________________
< Do what you like, like what you do. >
 -------------------------------------
        \   ^__^
         \  (oo)\_______
            (__)\       )\/\
                ||----w |
                ||     ||